bitluni/ESP32CameraI2S #3: Performance improvement for web server

derekmulcahy, 2017-12-15:
Hi,

I suggest writing the output to the client in one block instead of iterating.
```
client.write(bmpHeader, BMP::headerSize);
client.write(camera->frame, camera->xres * camera->yres * 2);
```
gave me a 6 frames per second instead of one frame every three seconds.

Cheers,
Derek

malloc32, 2018-05-08:
I use this code and the performance increase is amazing.
Thank you.

espiot328266, 2019-02-04:
hi thanks for improving the code
can you add more explain because i don't know how to edit the part of the code that improve the preference

espiot328266, 2019-02-04:
> I use this code and the performance increase is amazing.
> Thank you.

please help me to edit the program because i don't have any idea

malloc32, 2019-02-04:
[Here](https://github.com/bitluni/ESP32CameraI2S/pull/4/commits/e66900c916803173270184e7278c129176ebd00f) you can see the pull request made by derekmulcahy. You have to make these changes.

malloc32, 2019-02-04:
The file ESP32_I2S_Camera/ESP32_I2S_Camera.ino , about line 90 . You have to found where you have installed the library.

espiot328266, 2019-02-04:
> > The file ESP32_I2S_Camera/ESP32_I2S_Camera.ino , about line 90 . You have to found where you have installed the library.
> 
> i'm so stupid i guess
> because i can't understand what you mean
> can you make it easy for dummy me

i have done this 
`#include <WiFi.h>
#include <esp_wifi.h>
#include <lwip/err.h>
#include <lwip/sockets.h>
#include "BMP.h"
#include "OV7670.h"
#include "wifi_internal.h"
WiFiClient client;

const int SIOD = SDA;
const int SIOC = SCL;

const int VSYNC = 32;
const int HREF = 39;

const int XCLK = 18;
const int PCLK = 36;

const int D0 = 12;
const int D1 = 14;
const int D2 = 27;
const int D3 = 26;
const int D4 = 25;
const int D5 = 35;
const int D6 = 34;
const int D7 = 33;

#define SSID "Test"

#define MAX_PACKET_SIZE 1460
#define MAX_DATA_SIZE 1450

struct PacketHeader {
  uint8_t frame;
  uint32_t offset;
  uint32_t len;
};

OV7670 *camera;

byte bmpHeader[BMP::headerSize];
int udp_server = -1;
struct sockaddr_in destination;
bool video_running = false;

void setup() {
  Serial.begin(115200);

  WiFi.disconnect(true);

  // Force data rate
  wifi_internal_rate_t rate;
  rate.fix_rate = RATE_MCS4_SP;
  esp_wifi_internal_set_rate(100, 1, 4, &rate);

  WiFi.softAP(SSID);
  WiFi.onEvent(onClientChange, SYSTEM_EVENT_AP_STACONNECTED);
  WiFi.onEvent(onClientChange, SYSTEM_EVENT_AP_STADISCONNECTED);

  // Change beacon_interval because 100ms is crazy!
  wifi_config_t conf;
  esp_wifi_get_config(WIFI_IF_AP, &conf);
  conf.ap.beacon_interval = 3000;
  esp_wifi_set_config(WIFI_IF_AP, &conf);
  
  camera = new OV7670(OV7670::Mode::QQVGA_RGB565, SIOD, SIOC, VSYNC, HREF, XCLK, PCLK, D0, D1, D2, D3, D4, D5, D6, D7);
  BMP::construct16BitHeader(bmpHeader, camera->xres, camera->yres);

  // Init socket
  udp_server = socket(AF_INET, SOCK_DGRAM, 0);
  destination.sin_addr.s_addr = (uint32_t)IPAddress(255,255,255,255);
  destination.sin_family = AF_INET;
  destination.sin_port = htons(3333);

  Serial.println("Up, Waiting for clients");
}

void onClientChange(system_event_id_t event) {
  // Only start sending video after a client connects to avoid flooding
  // the channel when the client is attempting to connect.
  video_running = WiFi.softAPgetStationNum() != 0;
  if (video_running) {
    Serial.println("Video started!");
  } else {
    Serial.println("STOPPED");
  }
}

void loop() {
  if (!video_running) {
    client.write(bmpHeader, BMP::headerSize);
      client.write(camera->frame, camera->xres * camera->yres * 2);
    return;
  }

  camera->oneFrame();

  static byte packet[MAX_PACKET_SIZE];
  static uint8_t framenum = 0;
  PacketHeader* header = (PacketHeader*)packet;
  header->frame = framenum++;
  header->len = camera->frameBytes;

  for (int i = 0; i < camera->frameBytes; i += MAX_DATA_SIZE) {
    bool end = i + MAX_DATA_SIZE > camera->frameBytes;
    int len = end ? camera->frameBytes - i : MAX_DATA_SIZE;
      header->offset = i;
      memcpy(packet + sizeof(PacketHeader), camera->frame + i, len);
    int sent = sendto(
      udp_server,
      packet, sizeof(PacketHeader) + len,
      0,
      (struct sockaddr*) &destination, sizeof(destination)
    );
  }

  // FPS counter
  static unsigned int frames = 0;
  static unsigned long last = millis();
  ++frames;
  unsigned long now = millis();
  if (now - last > 1000) {
    Serial.println(frames);
    last = now;
    frames = 0;
  }
}`

akarsh98, 2019-02-04:
#include "OV7670.h"

#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library

#include <WiFi.h>
#include <WiFiMulti.h>
#include <WiFiClient.h>
#include "BMP.h"

const int SIOD = 21; //SDA
const int SIOC = 22; //SCL

const int VSYNC = 34;
const int HREF = 35;

const int XCLK = 32;
const int PCLK = 33;

const int D0 = 27;
const int D1 = 17;
const int D2 = 16;
const int D3 = 15;
const int D4 = 14;
const int D5 = 13;
const int D6 = 12;
const int D7 = 4;

const int TFT_DC = 2;
const int TFT_CS = 5;
//DIN <- MOSI 23
//CLK <- SCK 18

#define ssid1        "bina"
#define password1    "iamakarsh"
//#define ssid2        ""
//#define password2    ""

Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS,  TFT_DC, 0/*no reset*/);
OV7670 *camera;

WiFiMulti wifiMulti;
WiFiServer server(80);

unsigned char bmpHeader[BMP::headerSize];

void serve()
{
  WiFiClient client = server.available();
  if (client) 
  {
    //Serial.println("New Client.");
    String currentLine = "";
    while (client.connected()) 
    {
      if (client.available()) 
      {
        char c = client.read();
        //Serial.write(c);
        if (c == '\n') 
        {
          if (currentLine.length() == 0) 
          {
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();
            client.print(
              "<style>body{margin: 0}\nimg{height: 100%; width: auto}</style>"
              "[image]"
              "[image]");
            client.println();
            break;
          } 
          else 
          {
            currentLine = "";
          }
        } 
        else if (c != '\r') 
        {
          currentLine += c;
        }
        
        if(currentLine.endsWith("GET /camera"))
        {
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:image/bmp");
            client.println();
            
             client.write(bmpHeader, BMP::headerSize);
            client.write(camera->frame, camera->xres * camera->yres * 2);
        }
      }
    }
    // close the connection:
    client.stop();
    //Serial.println("Client Disconnected.");
  }  
}

void setup() 
{
  Serial.begin(115200);

  wifiMulti.addAP(ssid1, password1);
  //wifiMulti.addAP(ssid2, password2);
  Serial.println("Connecting Wifi...");
  if(wifiMulti.run() == WL_CONNECTED) {
      Serial.println("");
      Serial.println("WiFi connected");
      Serial.println("IP address: ");
      Serial.println(WiFi.localIP());
  }
  
  camera = new OV7670(OV7670::Mode::QQVGA_RGB565, SIOD, SIOC, VSYNC, HREF, XCLK, PCLK, D0, D1, D2, D3, D4, D5, D6, D7);
  BMP::construct16BitHeader(bmpHeader, camera->xres, camera->yres);
  
  tft.initR(INITR_BLACKTAB);
  tft.fillScreen(0);
  server.begin();
}

void displayY8(unsigned char * frame, int xres, int yres)
{
  tft.setAddrWindow(0, 0, yres - 1, xres - 1);
  int i = 0;
  for(int x = 0; x < xres; x++)
    for(int y = 0; y < yres; y++)
    {
      i = y * xres + x;
      unsigned char c = frame[i];
      unsigned short r = c >> 3;
      unsigned short g = c >> 2;
      unsigned short b = c >> 3;
      tft.pushColor(r << 11 | g << 5 | b);
    }  
}

void displayRGB565(unsigned char * frame, int xres, int yres)
{
  tft.setAddrWindow(0, 0, yres - 1, xres - 1);
  int i = 0;
  for(int x = 0; x < xres; x++)
    for(int y = 0; y < yres; y++)
    {
      i = (y * xres + x) << 1;
      tft.pushColor((frame[i] | (frame[i+1] << 8)));
    }  
}

void loop()
{
  camera->oneFrame();
  serve();
  displayRGB565(camera->frame, camera->xres, camera->yres);
}

akarsh98, 2019-02-04:
> i just need code that work well is too complected to get all this

#include "OV7670.h"

#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library

#include <WiFi.h>
#include <WiFiMulti.h>
#include <WiFiClient.h>
#include "BMP.h"

const int SIOD = 21; //SDA
const int SIOC = 22; //SCL

const int VSYNC = 34;
const int HREF = 35;

const int XCLK = 32;
const int PCLK = 33;

const int D0 = 27;
const int D1 = 17;
const int D2 = 16;
const int D3 = 15;
const int D4 = 14;
const int D5 = 13;
const int D6 = 12;
const int D7 = 4;

const int TFT_DC = 2;
const int TFT_CS = 5;
//DIN <- MOSI 23
//CLK <- SCK 18

#define ssid1        "ssid"
#define password1    "pass"
//#define ssid2        ""
//#define password2    ""

Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS,  TFT_DC, 0/*no reset*/);
OV7670 *camera;

WiFiMulti wifiMulti;
WiFiServer server(80);

unsigned char bmpHeader[BMP::headerSize];

void serve()
{
  WiFiClient client = server.available();
  if (client) 
  {
    //Serial.println("New Client.");
    String currentLine = "";
    while (client.connected()) 
    {
      if (client.available()) 
      {
        char c = client.read();
        //Serial.write(c);
        if (c == '\n') 
        {
          if (currentLine.length() == 0) 
          {
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();
            client.print(
              "<style>body{margin: 0}\nimg{height: 100%; width: auto}</style>"
              "[image]"
              "[image]");
            client.println();
            break;
          } 
          else 
          {
            currentLine = "";
          }
        } 
        else if (c != '\r') 
        {
          currentLine += c;
        }
        
        if(currentLine.endsWith("GET /camera"))
        {
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:image/bmp");
            client.println();
            
             client.write(bmpHeader, BMP::headerSize);
            client.write(camera->frame, camera->xres * camera->yres * 2);
        }
      }
    }
    // close the connection:
    client.stop();
    //Serial.println("Client Disconnected.");
  }  
}

void setup() 
{
  Serial.begin(115200);

  wifiMulti.addAP(ssid1, password1);
  //wifiMulti.addAP(ssid2, password2);
  Serial.println("Connecting Wifi...");
  if(wifiMulti.run() == WL_CONNECTED) {
      Serial.println("");
      Serial.println("WiFi connected");
      Serial.println("IP address: ");
      Serial.println(WiFi.localIP());
  }
  
  camera = new OV7670(OV7670::Mode::QQVGA_RGB565, SIOD, SIOC, VSYNC, HREF, XCLK, PCLK, D0, D1, D2, D3, D4, D5, D6, D7);
  BMP::construct16BitHeader(bmpHeader, camera->xres, camera->yres);
  
  tft.initR(INITR_BLACKTAB);
  tft.fillScreen(0);
  server.begin();
}

void displayY8(unsigned char * frame, int xres, int yres)
{
  tft.setAddrWindow(0, 0, yres - 1, xres - 1);
  int i = 0;
  for(int x = 0; x < xres; x++)
    for(int y = 0; y < yres; y++)
    {
      i = y * xres + x;
      unsigned char c = frame[i];
      unsigned short r = c >> 3;
      unsigned short g = c >> 2;
      unsigned short b = c >> 3;
      tft.pushColor(r << 11 | g << 5 | b);
    }  
}

void displayRGB565(unsigned char * frame, int xres, int yres)
{
  tft.setAddrWindow(0, 0, yres - 1, xres - 1);
  int i = 0;
  for(int x = 0; x < xres; x++)
    for(int y = 0; y < yres; y++)
    {
      i = (y * xres + x) << 1;
      tft.pushColor((frame[i] | (frame[i+1] << 8)));
    }  
}

void loop()
{
  camera->oneFrame();
  serve();
  displayRGB565(camera->frame, camera->xres, camera->yres);
}

espiot328266, 2019-02-04:
> #include "OV7670.h"
>
> #include <Adafruit_GFX.h> // Core graphics library
> #include <Adafruit_ST7735.h> // Hardware-specific library
>
> #include <WiFi.h>
> #include <WiFiMulti.h>
> #include <WiFiClient.h> [...]

thank you so much for your attention 
but the code is give me this errour 
Arduino: 1.8.9 Hourly Build 2019/01/24 07:33 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default, 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Users\pc\Downloads\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\pc\Downloads\arduino-nightly\hardware -hardware C:\Users\pc\AppData\Local\Arduino15\packages -tools C:\Users\pc\Downloads\arduino-nightly\tools-builder -tools C:\Users\pc\Downloads\arduino-nightly\hardware\tools\avr -tools C:\Users\pc\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\pc\Downloads\arduino-nightly\libraries -libraries C:\Users\pc\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path C:\Users\pc\AppData\Local\Temp\arduino_build_963327 -warnings=none -build-cache C:\Users\pc\AppData\Local\Temp\arduino_cache_610903 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.0 -prefs=runtime.tools.esptool_py-2.6.0.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.0 -verbose C:\Users\pc\Desktop\ESP32CameraI2S-master - Copy\hadi_jat_hmd\hadi_jat_hmd.ino
C:\Users\pc\Downloads\arduino-nightly\arduino-builder -compile -logger=machine -hardware C:\Users\pc\Downloads\arduino-nightly\hardware -hardware C:\Users\pc\AppData\Local\Arduino15\packages -tools C:\Users\pc\Downloads\arduino-nightly\tools-builder -tools C:\Users\pc\Downloads\arduino-nightly\hardware\tools\avr -tools C:\Users\pc\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\pc\Downloads\arduino-nightly\libraries -libraries C:\Users\pc\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10809 -build-path C:\Users\pc\AppData\Local\Temp\arduino_build_963327 -warnings=none -build-cache C:\Users\pc\AppData\Local\Temp\arduino_cache_610903 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.0 -prefs=runtime.tools.esptool_py-2.6.0.path=C:\Users\pc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.0 -verbose C:\Users\pc\Desktop\ESP32CameraI2S-master - Copy\hadi_jat_hmd\hadi_jat_hmd.ino
Using board 'esp32' from platform in folder: C:\Users\pc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1
Using core 'esp32' from platform in folder: C:\Users\pc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1
Detecting libraries used...
"C:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/config" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/app_trace" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/app_update" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/asio" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/bootloader_support" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/bt" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/coap" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/console" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/driver" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/esp-tls" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\pac [... cut ...] l\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/asio" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/bootloader_support" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/bt" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/coap" "-IC:\\Users\\pc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.

espiot328266, 2019-02-04:
> #include "OV7670.h"
>
> #include <Adafruit_GFX.h> // Core graphics library
> #include <Adafruit_ST7735.h> // Hardware-specific library
>
> #include <WiFi.h>
> #include <WiFiMulti.h>
> #include <WiFiClient.h> [...]

please reply

espiot328266, 2019-02-04:
i don't need to use the tft or mqtt
because basically i need to use the webserver to show date 
can i remove the tft in the code dose it will work like that 
because i did what he said in the video and comented them and is compiling

espiot328266, 2019-02-05:
@malloc32 @derekmulcahy @akarsh98 
can i not use the tft 
because i want to access to camera jus with webserver 
like this code 
is compile
 #include "OV7670.h"

#include <Adafruit_GFX.h> // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library

#include <WiFi.h>
#include <WiFiMulti.h>
#include <WiFiClient.h>
#include "BMP.h"

const int SIOD = 21; //SDA
const int SIOC = 22; //SCL

const int VSYNC = 34;
const int HREF = 35;

const int XCLK = 32;
const int PCLK = 33;

const int D0 = 27;
const int D1 = 17;
const int D2 = 16;
const int D3 = 15;
const int D4 = 14;
const int D5 = 13;
const int D6 = 12;
const int D7 = 4;

const int TFT_DC = 2;
const int TFT_CS = 5;
//DIN <- MOSI 23
//CLK <- SCK 18

#define ssid1 "ssid"
#define password1 "pass"
//#define ssid2 ""
//#define password2 ""

//Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, 0/no reset/);
OV7670 *camera;

WiFiMulti wifiMulti;
WiFiServer server(80);

unsigned char bmpHeader[BMP::headerSize];

void serve()
{
WiFiClient client = server.available();
if (client)
{
//Serial.println("New Client.");
String currentLine = "";
while (client.connected())
{
if (client.available())
{
char c = client.read();
//Serial.write(c);
if (c == '\n')
{
if (currentLine.length() == 0)
{
client.println("HTTP/1.1 200 OK");
client.println("Content-type:text/html");
client.println();
client.print(
"<style>body{margin: 0}\nimg{height: 100%; width: auto}</style>"
""
"");
client.println();
break;
}
else
{
currentLine = "";
}
}
else if (c != '\r')
{
currentLine += c;
}

    if(currentLine.endsWith("GET /camera"))
    {
        client.println("HTTP/1.1 200 OK");
        client.println("Content-type:image/bmp");
        client.println();
        
         client.write(bmpHeader, BMP::headerSize);
        client.write(camera->frame, camera->xres * camera->yres * 2);
    }
  }
}
// close the connection:
client.stop();
//Serial.println("Client Disconnected.");
}
}

void setup()
{
Serial.begin(115200);

wifiMulti.addAP(ssid1, password1);
//wifiMulti.addAP(ssid2, password2);
Serial.println("Connecting Wifi...");
if(wifiMulti.run() == WL_CONNECTED) {
Serial.println("");
Serial.println("WiFi connected");
Serial.println("IP address: ");
Serial.println(WiFi.localIP());
}

camera = new OV7670(OV7670::Mode::QQVGA_RGB565, SIOD, SIOC, VSYNC, HREF, XCLK, PCLK, D0, D1, D2, D3, D4, D5, D6, D7);
BMP::construct16BitHeader(bmpHeader, camera->xres, camera->yres);

//tft.initR(INITR_BLACKTAB);
//tft.fillScreen(0);
server.begin();
}

void displayY8(unsigned char * frame, int xres, int yres)
{
//tft.setAddrWindow(0, 0, yres - 1, xres - 1);
int i = 0;
for(int x = 0; x < xres; x++)
for(int y = 0; y < yres; y++)
{
i = y * xres + x;
unsigned char c = frame[i];
unsigned short r = c >> 3;
unsigned short g = c >> 2;
unsigned short b = c >> 3;
//tft.pushColor(r << 11 | g << 5 | b);
}
}

void displayRGB565(unsigned char * frame, int xres, int yres)
{
//tft.setAddrWindow(0, 0, yres - 1, xres - 1);
int i = 0;
for(int x = 0; x < xres; x++)
for(int y = 0; y < yres; y++)
{
i = (y * xres + x) << 1;
//tft.pushColor((frame[i] | (frame[i+1] << 8)));
}
}

void loop()
{
camera->oneFrame();
serve();
displayRGB565(camera->frame, camera->xres, camera->yres);
}

malloc32, 2019-02-05:
ok, then now it is working?, Some time ago I leave to program microcontrollers, no time :(.

espiot328266, 2019-02-05:
i have compared the code you gave me and the original code 
i found that you are missing somethings 
and i have added it now it have compiled with not much change 

just he one you have gave me 
thank you all for the help and attention 

`#include "OV7670.h"

#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library

#include <WiFi.h>
#include <WiFiMulti.h>
#include <WiFiClient.h>
#include "BMP.h"

const int SIOD = 21; //SDA
const int SIOC = 22; //SCL

const int VSYNC = 34;
const int HREF = 35;

const int XCLK = 32;
const int PCLK = 33;

const int D0 = 27;
const int D1 = 17;
const int D2 = 16;
const int D3 = 15;
const int D4 = 14;
const int D5 = 13;
const int D6 = 12;
const int D7 = 4;

const int TFT_DC = 2;
const int TFT_CS = 5;
//DIN <- MOSI 23
//CLK <- SCK 18

#define ssid1        "YOUR_WIFI_SSID"
#define password1    "YOUR_PASSWORD"
//#define ssid2        ""
//#define password2    ""

Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS,  TFT_DC, 0/*no reset*/);
OV7670 *camera;

WiFiMulti wifiMulti;
WiFiServer server(80);

unsigned char bmpHeader[BMP::headerSize];

void serve()
{
  WiFiClient client = server.available();
  if (client) 
  {
    //Serial.println("New Client.");
    String currentLine = "";
    while (client.connected()) 
    {
      if (client.available()) 
      {
        char c = client.read();
        //Serial.write(c);
        if (c == '\n') 
        {
          if (currentLine.length() == 0) 
          {
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();
            client.print(
              "<style>body{margin: 0}\nimg{height: 100%; width: auto}</style>"
              "[image]"
              "[image]");
            client.println();
            break;
          } 
          else 
          {
            currentLine = "";
          }
        } 
        else if (c != '\r') 
        {
          currentLine += c;
        }
        
        if(currentLine.endsWith("GET /camera"))
        {
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:image/bmp");
            client.println();
            
            client.write(bmpHeader, BMP::headerSize);
            client.write(camera->frame, camera->xres * camera->yres * 2);
        }
      }
    }
    // close the connection:
    client.stop();
    //Serial.println("Client Disconnected.");
  }  
}

void setup() 
{
  Serial.begin(115200);

  wifiMulti.addAP(ssid1, password1);
  //wifiMulti.addAP(ssid2, password2);
  Serial.println("Connecting Wifi...");
  if(wifiMulti.run() == WL_CONNECTED) {
      Serial.println("");
      Serial.println("WiFi connected");
      Serial.println("IP address: ");
      Serial.println(WiFi.localIP());
  }
  
  camera = new OV7670(OV7670::Mode::QQVGA_RGB565, SIOD, SIOC, VSYNC, HREF, XCLK, PCLK, D0, D1, D2, D3, D4, D5, D6, D7);
  BMP::construct16BitHeader(bmpHeader, camera->xres, camera->yres);
  
  tft.initR(INITR_BLACKTAB);
  tft.fillScreen(0);
  server.begin();
}

void displayY8(unsigned char * frame, int xres, int yres)
{
  tft.setAddrWindow(0, 0, yres - 1, xres - 1);
  int i = 0;
  for(int x = 0; x < xres; x++)
    for(int y = 0; y < yres; y++)
    {
      i = y * xres + x;
      unsigned char c = frame[i];
      unsigned short r = c >> 3;
      unsigned short g = c >> 2;
      unsigned short b = c >> 3;
      tft.pushColor(r << 11 | g << 5 | b);
    }  
}

void displayRGB565(unsigned char * frame, int xres, int yres)
{
  tft.setAddrWindow(0, 0, yres - 1, xres - 1);
  int i = 0;
  for(int x = 0; x < xres; x++)
    for(int y = 0; y < yres; y++)
    {
      i = (y * xres + x) << 1;
      tft.pushColor((frame[i] | (frame[i+1] << 8)));
    }  
}

void loop()
{
  camera->oneFrame();
  serve();
  displayRGB565(camera->frame, camera->xres, camera->yres);
}`

espiot328266, 2019-02-06:
the code now worked but there is a problem the image dose not give me good colors 
and i have changed pin 17 with pin 25 because i  can't find pin 17 
do yo have any ideas

malloc32, 2019-02-06:
ufff, sorry It was many time ago, even now I have not the schema..., but I remember that I have good colors, not very good image quality because the resolution.. and the price of the camera... I cost about 2€ in aliexpress, so....

espiot328266, 2019-02-06:
the colors i got is not good at all i berly can see my face 
i have esp32 wroom and it dose not have pin 17 
how can i change the pin with no problem 
there the esp i have 
[image]
 and i want to know how the chnage the screen size so i can see it full using the phone

espiot328266, 2019-02-06:
i  have 17 pin is tx2 but now when i changed the router it dose not show the ip addressee of the esp in serial monitor

espiot328266, 2019-02-07:
@derekmulcahy  @malloc32 
i need to add code led control in the camera code
any help?

did this code but i need to optimized it 
#include "OV7670.h"

#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library

#include <WiFi.h>
#include <WiFiMulti.h>
#include <WiFiClient.h>
#include "BMP.h"

String header;

// Auxiliar variables to store the current output state
String output22State = "off";

// Assign output variables to GPIO pins
const int D25 = 25;
String output25State = "off";
const int SIOD = 21; //SDA
const int SIOC = 22; //SCL

const int VSYNC = 34;
const int HREF = 35;

const int XCLK = 32;
const int PCLK = 33;

const int D0 = 27;
const int D1 = 17;
const int D2 = 16;
const int D3 = 15;
const int D4 = 14;
const int D5 = 13;
const int D6 = 12;
const int D7 = 4;

const int TFT_DC = 2;
const int TFT_CS = 5;
//DIN <- MOSI 23
//CLK <- SCK 18

#define ssid1        "YOUR_WIFI_SSID"
#define password1    "YOUR_PASSWORD"
//#define ssid2        ""
//#define password2    ""

Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS,  TFT_DC, 0/*no reset*/);
OV7670 *camera;

WiFiMulti wifiMulti;
WiFiServer server(80);

unsigned char bmpHeader[BMP::headerSize];

void serve()
{
  WiFiClient client = server.available();
  if (client) 
  {
    //Serial.println("New Client.");
    String currentLine = "";
    while (client.connected()) 
    {
      if (client.available()) 
      {
[... 152 lines cut ...]
              client.println("<p><a href=\"/25/off\"><button class=\"button button2\">OFF</button></a></p>");
            }     
             client.println("</body></html>");
            
            // The HTTP response ends with another blank line
            client.println();
            // Break out of the while loop
            break;
          } else { // if you got a newline, then clear currentLine
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }
      }
    }
    // Clear the header variable
    header = "";
    // Close the connection
    client.stop();
    Serial.println("Client disconnected.");
    Serial.println("");
  }
}
}

derekmulcahy, 2019-02-08:
Thanks for the merge!

closed 2019-02-08 by derekmulcahy