bitluni/ESP32CameraI2S #11: It is not compiling

shikharuniyal, 2019-01-12:
when i try to compile the code it is showing this type of error:-

**C:\Users\SHIKHAR\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp: In member function 'virtual void Adafruit_ST7735::setRotation(uint8_t)':

C:\Users\SHIKHAR\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:292:6: error: '_xstart' was not declared in this scope

      _xstart   = _colstart;

      ^

C:\Users\SHIKHAR\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:293:6: error: '_ystart' was not declared in this scope

      _ystart   = _rowstart;

      ^

exit status 1
Error compiling for board WEMOS LOLIN32.**

can anyone give a sol. for this error.

shikharuniyal, 2019-01-12:
when i updated the driver it is showing 

**_In file included from sketch\I2SCamera.h:17:0,

                 from sketch\OV7670.h:2,

                 from C:\Users\SHIKHAR\Desktop\ESP32CameraI2S-master\ESP32_I2S_Camera\ESP32_I2S_Camera.ino:15:

DMABuffer.h:2: error: 'malloc' does not name a type

 malloc

 ^

In file included from sketch\OV7670.h:2:0,

                 from C:\Users\SHIKHAR\Desktop\ESP32CameraI2S-master\ESP32_I2S_Camera\ESP32_I2S_Camera.ino:15:

I2SCamera.h:31: error: 'DMABuffer' does not name a type

   static DMABuffer **dmaBuffer;

          ^

exit status 1
'malloc' does not name a type_**

please help me out!!!!!

johannes1999, 2019-01-12:
When i try to compile i get a similar error:
"
In file included from sketch\I2SCamera.h:17:0,

                 from sketch\OV7670.h:2,

                 from sketch\OV7670.cpp:1:

sketch\DMABuffer.h: In constructor 'DMABuffer::DMABuffer(int)':

DMABuffer.h:10:43: error: 'malloc' was not declared in this scope

     buffer = (unsigned char *)malloc(bytes);

                                           ^

In file included from sketch\I2SCamera.h:17:0,

                 from sketch\I2SCamera.cpp:6:

sketch\DMABuffer.h: In constructor 'DMABuffer::DMABuffer(int)':

DMABuffer.h:10:43: error: 'malloc' was not declared in this scope

     buffer = (unsigned char *)malloc(bytes);

                                           ^
"

When I downgraded the esp32 version in the boardmanager from 1.0.1 to 1.0.0 then it compiled, but the programm does not run an gave me that message over the serial interface:
"E (3646) ledc: ledc_set_duty_with_hpoint(383): hpoint argument is invalid"

I have also tried to downgrade the IDE to version 1.8.4 which bitluni used, but that made no difference.

shikharuniyal, 2019-01-13:
i again tried with newer version and the results are same 
In file included from sketch\I2SCamera.h:17:0,

                 from sketch\I2SCamera.cpp:6:

sketch\DMABuffer.h: In constructor 'DMABuffer::DMABuffer(int)':

DMABuffer.h:10: error: 'malloc' was not declared in this scope

     buffer = (unsigned char *)malloc(bytes);

                                           ^

exit status 1
'malloc' was not declared in this scope
   
i think that might including a library for this may correct this error

shikharuniyal, 2019-01-13:
i had tried to use this malloc.h from https://github.com/esp8266/Arduino/blob/master/tools/sdk/libc/xtensa-lx106-elf/include/malloc.h in a new tab and now it shows
Arduino: 1.8.2 (Windows XP), Board: "WEMOS LOLIN32, 80MHz, Default, 921600"

Traceback (most recent call last):

  File "esptool.py", line 35, in <module>

  File "c:\python27\Lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 389, in load_module

  File "site-packages\serial\__init__.py", line 27, in <module>

  File "c:\python27\Lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 389, in load_module

  File "site-packages\serial\serialwin32.py", line 15, in <module>

  File "c:\python27\Lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 389, in load_module

  File "site-packages\serial\win32.py", line 182, in <module>

  File "ctypes\__init__.py", line 375, in __getattr__

  File "ctypes\__init__.py", line 380, in __getitem__

AttributeError: function 'CancelIoEx' not found

Failed to execute script esptool

exit status -1
Error compiling for board WEMOS LOLIN32.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

shikharuniyal, 2019-01-13:
do we have to put a int  value in place of bytes if it is so then which value
class DMABuffer
{
  public:
  lldesc_t descriptor;
  unsigned char* buffer;
  DMABuffer(int bytes)
  {
    buffer = (unsigned char *)malloc(bytes);
    descriptor.length = bytes;
    descriptor.size = descriptor.length;
    descriptor.owner = 1;
    descriptor.sosf = 1;
    descriptor.buf = (uint8_t*) buffer;
    descriptor.offset = 0;
    descriptor.empty = 0;
    descriptor.eof = 1;
    descriptor.qe.stqe_next = 0;
  }

Samlof, 2019-01-13:
Going back to 1.0.0 ESP32 and compiling works. Then you have to add
	"ch_conf.hpoint = 0;"  into XClk.cpp before "err = ledc_channel_config(&ch_conf);". This is where I left off and not working yet. Might be problems with my own pin setup, or i2s has changed.

shikharuniyal, 2019-01-17:
still the results are same 

In file included from sketch\I2SCamera.h:17:0,

                 from sketch\I2SCamera.cpp:6:

sketch\DMABuffer.h: In constructor 'DMABuffer::DMABuffer(int)':

DMABuffer.h:10: error: 'malloc' was not declared in this scope

     buffer = (unsigned char *)malloc(bytes);

                                           ^
exit status 1
'malloc' was not declared in this scope

pmpz, 2019-03-08:
just add 
#include <stdlib.h> 
in DMABuffer.h

shikharuniyal, 2019-03-18:
thanks now it works

closed 2019-03-18 by shikharuniyal

sirmagid, 2019-06-26:
nice work