bitluni/ESP32CameraI2S #13: Leaving... Hard resetting via RTS pin...

cpcpcpcpcpc, 2019-03-09:
Hi, I used to be able to make the camera work, but since few week, some updates on my Arduino or Hardware library ESP32 generated the following message after compiling and uploading to the ESP32 Lolin32 board in the Arduino IDE bottom window:

Leaving...
Hard resetting via RTS pin...

Have you guys experienced the same thing?
The main problem being that now the camera does not work anymore and I also get in the serial monitor the following message:
E (2499) ledc: ledc_set_duty_with_hpoint(383): hpoint argument is invalid

Can you please help?

thanks

shikharuniyal, 2019-03-18:
when i use ch_conf.hpoint = 0;
then it shows like this  -->

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

can anyone give me suggestions.

s0-er-eN, 2019-04-04:
### +1 
I'm having the same issue. I also set hpoint to 0 before calling `ledc_channel_config()` in XClk.cpp but the image shown on my tft display is the following: 
[image]

anyone had a similar issue so far and is willing to help?