bitluni/ESP32CameraI2S #17: OV7670::Mode::VGA_RGB565 not working

guipiveti, 2019-07-02:
When I change the mode to OV7670::Mode::VGA_RGB565 (camera = new OV7670(OV7670::Mode::VGA_RGB565, SIOD, SIOC, VSYNC, HREF, XCLK, PCLK, D0, D1, D2, D3, D4, D5, D6, D7);) 
It results in: Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
It look like it happens when dmaBuffer[0]->sampleCount() is called in I2SCamera.cpp

GDanovski, 2019-07-02:
Hello! Everything is working well if the mode is QQVGA_RGB565 or QQQVGA_RGB565. It fails if the resolution of the obtained image is set to higher than that. I thing that the ESP32 is not able to process bigger images (low memory or something else, not sure).

Emb-Coder, 2022-08-13:
I got the Same issue for Esp32 Dev module. ---Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled---. if I chose higher resolution. Max resolution working is QQVGA_RGB565. any suggestion ?