bitluni/ESP32CameraI2S #19: Getting Issue in Grayscale image

ashtam55, 2020-01-21:
Hey Sir,
Thanks for this wonderful library which works flawlessly. It's just I want to get monochrome output from the camera module ov7670 which I can't able to get. Please let me know how to do it in ESP32.

bitluni, 2020-01-21:
ooh that is very long ago. did you try any of my code? the osci display
code might use b/w, I don't really remember

ashtam55, 2020-01-21:
Thanks for replying Sir, 
I am working with your code want to change the format in YUV which I achieved by changing the COM7 register but now I need the image in grayscale for which I need to discard U and V components from the frame as I just need Y. There is not enough documentation to work around for esp32_arduino.