bitluni/ESP32CameraI2S #25: ESP32 WROOM 32 with OV7670 no FIFO

antoniojpmarin, 2024-01-24:
Hello! I've a problem with the code. The fact is that I get this kind of images from my camera:
[image]

I've used this byte sequence: SM_0A0B_0C0D

How can I solve it??

snw0502, 2024-03-24:
> Hello! I've a problem with the code. The fact is that I get this kind of images from my camera: [...]

did u ever figure out the issue here? i am having the same

GastonMelo, 2024-10-22:
Im using the same board and getting images like that. I just simply trying to get the bar test from the camera. Did you solve it?
Gastón